iFallUpHill/flow-calculator #5: gcode assumes relative extrusion

strayr, 2022-09-07:
Should probably have a warning about this, or specify explicitly AFTER the start gcode?

Yeah, it's in the start gcode, but not if you use PRINT_START and didn't bother reading the example gcode.

iFallUpHill, 2022-09-08:
That's a good point and a great catch, can you think of a reason to not force an M83 after the start Gcode?

strayr, 2022-09-09:
Unless there's a firmware that it crashes, no, do this explicity. I suspect that many users will copypasta their known good gcode without reading so I'd be inclined to force the required behaviour, otherwise it's a point of failure you don't need.

iFallUpHill, 2022-09-18:
Sounds good to me, added!

closed 2022-09-18 by iFallUpHill

strayr, 2022-10-11:
Unless there's firmware that rejects it and halts, no.

I'm used to doing klipper macros that move steppers inside a saved gcode
state, precisely because relying on the printer being in a known state is
dangerous. If I still had a collection of OEM supplied marlin running on my
printers I'd probably be even more paranoid.