iFallUpHill/flow-calculator #6: flow doesn't reset if temperature step is zero

strayr, 2022-10-05:
I think this is a docs issue, and its actually a feature I like.

If is set a temperature step of zero, it just carries on the next row, this is great, I can get 15 steps and it's actually really useful, but waht I was looking for was 3 sets at identical temperature. Not a problem, just needs to be documented somehow

strayr, 2022-10-05:
If i set temperature offset of zero and a temperature step of 1 get one column, despite "inclusive; Set to 1 to fill plate with flow steps instead"

iFallUpHill, 2022-11-06:
@strayr apologies that it took a while for me to get back to this, but just to make sure that I understand:

You're looking for a mode where you can repeat the same column `n` times (i.e. repeating the same flow step test `n` times)?

strayr, 2022-11-06:
Initially I was looking for no change in temperature and increasing flow across rows and columns, which is possible but the text saying how to get that is unclear and confusing. 

A mode repeating the same column n times would be nice but I ended up approximating that by having a really tiny temperature step.

What I tend to find myself doing is printing 5mm3-20mm3 in 1mm3 steps on one plate (double those numbers for a high flow set-up) and then 5 steps across the "knee" three times

iFallUpHill, 2022-11-27:
Oh, got it. I'll think about how to describe that feature better. Glad to know that it's working as intended (though unclear how we got there). 

For context, when I was initially porting the spreadsheet version over, I copied all of the terminology 1:1 with the assumption that people would land on the tool after watching the original video / reading the original repo. I realize this is not the case now, so thank you for helping me understand the confusion!

strayr, 2022-11-27:
That's where I started from, but I'm looking to drill down on the curve
where underextrusion starts.