iFallUpHill/flow-calculator #7: Suggestion: For Delta printers, support locations at negative X and Y coordinates

progman32, 2022-10-09:
Many (all?) delta configuration printers treat 0,0 as the center of the print area, with coordinates going both negative and positive in X and Y. The tool currently assumes 0,0 is the machine's limit, which makes 3/4 of the bed of these printers inaccessible to the tool. Perhaps two new text areas could be added: min X and min Y.

Extra bonus points for being able to arrange the test points in a circle :)

progman32, 2022-10-09:
Side note: the current "valid be size" check doesn't let me put in the halved value of my delta's bed size, since it enforces a minimum of 100.

iFallUpHill, 2022-10-31:
I'll have to think about this a bit more as it would be neat to be able to take full advantage of the entire build area. I'm not entirely sure if the math checks out for computing with negative bed dimensions though, so if it doesn't, it might take a while to get to this.

iFallUpHill, 2022-11-27:
In the interim, I bumped the min bed length requirement down to 60mm.

AudreyAP, 2023-04-28:
There is a very easy workaround for this! I'm using Klipper firmware but this should work for Marlin as well - I've not tested marlin at all. 

# PLEASE USE A SLOW TRAVEL SPEED THE FIRST TIME YOU TRY THIS! I'm not responsible if you damage your printer, etc. 

[image]

The outer circle is the safe printing area of my bed. The inner square was adjusted to fit nicely inside - I picked 180mm because it's a nice number. 

Once you've determined the edge length of a square that will fit inside your bed, set the bed size in the generator:

[image]

Next, divide the edge length by two. In my case, that would be 90mm. Adjust your start and end gcode:

Start gcode:
```
YOUR NORMAL START GCODE HERE

G0 X0 Y0 Z50 F600 ; Move to origin - do not adjust this

G92 X90 Y90 ; Apply gcode offset - ADJUST THIS NUMBER TO MATCH YOUR PRINTER
```

End gcode:
```
G28 ; Home to reset origin due to delta workaround
```

You will need to manually turn heaters off, etc. End gcode varies wildly from person to person, I can't provide something that will work for everyone. 

Export and run the gcode and everything should work!

I used Fusion360 to do the calculations for this but you could do it on graph paper if you want, even.