iFallUpHill/flow-calculator #15: no blobs but prime

TylonHH, 2023-04-24:
Hey,
I run my second test and raised the amount too 400 and increased the height to 15mm
But it wont print it. The starting feedrate is lower then in the 1st test.

Is in my config something wrong or is this an firmware limitation?

Here are my setting:
```
;####### Settings
; bedWidth = 330
; bedLength = 200
; bedMargin = 10
; safeZPark = 10
; filamentDiameter = 1.75
; travelSpeed = 100
; dwellTime = 10
; bedTemp = 60
; fanSpeed = 0
; primeLength = 25
; primeAmount = 20
; primeSpeed = 5
; wipeLength = 15
; retractionDistance = 0.8
; retractionSpeed = 35
; blobHeight = 15
; extrusionAmount = 400
; direction = 1
; flowSpacing = 35
; tempSpacing = 38
; flowStart = 16
; flowOffset = 2
; flowSteps = 6
; flowEnd = 26
; tempStart = 210
; tempOffset = 20
; tempSteps = 3
; tempEnd = 250
```

and this is the 1st drop
```
;####### 210°C // 16mm3/s
M117 210C // 16mm3/s
G0 X10 Y10 Z20.5 F6000
G4 S10 ; Dwell
G0 Z0.3 ; Drop down
G1 X35 E20 F300 ; Prime
G1 E-0.8 F2100 ; Retract
G0 X50 F6000 ; Wipe
G0 Z0.5 ; Lift
G1 E0.8 F2100 ; Undo Retract
G1 Z15.5 E400 F14.97 ; Extrude
G1 E-0.8 F2100 ; Retract
G0 Z20.5; Lift
G0 X10 Y10 F6000
G92 E0 ; Reset Extruder
```

TylonHH, 2023-04-24:
[image]

TylonHH, 2023-04-24:
if I change the amount to 200mm and height to 10mm it works. strange

iFallUpHill, 2023-05-22:
Were you able to figure this out?

TylonHH, 2023-05-22:
> Were you able to figure this out?
Nope. I guess there is a firmware limiter or something. As I decreased the value it worked.

gtorelly, 2023-06-08:
I had the same issue. Any value above 200 mm would fail. My printer uses Marlin firmware and there is a variable that limits the maximum length of a single extrude in "Configuration.h":
`/**
 * Prevent a single extrusion longer than EXTRUDE_MAXLENGTH.
 * Note: For Bowden Extruders make this large enough to allow load/unload.
 */
#define PREVENT_LENGTHY_EXTRUDE
#define EXTRUDE_MAXLENGTH 1000`

I increased it to 1000 and now the blob test works with longer lengths (up to 1000)