iFallUpHill/flow-calculator #22: delta printers / negative offsets

tmbdev, 2023-07-23:
It's hard to generate G-code for delta printers with the current system. It would be nice to at least have the option of having negative offsets to fit a square test pattern into the center of a delta printer.

Ideal, of course, would be adding a circular/polar coordinate pattern for round print beds, with flow rates corresponding to different angles and temperatures corresponding to different radii.

Doct0rDuc, 2023-08-11:
yes it would be really handy, for now I'm using standard flow values ​​taken from the internet for the hotend  I have on my delta, but I'd like to have more accurate and precise ones for each type of filament I use

mthqwork, 2023-11-18:
+1, it'd be great to have that.